dontsovcmc/waterius #198: Список платформ автоматизации ТСЖ, УК

dontsovcmc, 2021-09-23:
После 60-го сайта интегрированного в waterius.ru, а также пары запросов от жильцов "какие есть кабинеты, чтобы предложить председателям", понял, пришла пора собирать список компаний.

Т.к. я не специалист, то не знаю, какие характеристики вытащить в список. Добавляйте в комментариях, я обновлю.

https://stack-it.ru
г. Ярославль

https://sm-center.ru
г.Барнаул

https://billing74.ru
г. Челябинск

ostap7474, 2021-10-15:
Не могли бы вы добавить для г.Уфы  МУП ЕРКЦ https://erkcufa.ru

dontsovcmc, 2022-01-19:
@ostap7474 с https://erkcufa.ru есть технические сложности. он на древней технологии ASP сделан с защитой, поэтому вынуждены отложить интеграцию. А через приложение для телефона показания отправляются? Очень низкий рейтинг у него.

ostap7474, 2022-01-24:
Установил приложение на мобильный, показания отправил. Работает вроде
нормально. Что, нет никакой надежды в ближайшее время? Можно продавать
Ватериус?

ostap7474, 2022-01-24:
Через небольшой промежуток времени показания которые передал через
мобильное приложение появились на сайте.

dontsovcmc, 2022-08-04:
@ostap7474 добавили передачу в Уфу! Пользуйтесь на здоровье =)

andrewsan2002, 2023-03-30:
Добрый день. Можете добавить передачу в https://rkc-m.ru/ ?

Crazy-Child, 2023-04-09:
Добавьте пожалуйста в https://kabinet.uk/ Челябинск

dontsovcmc, 2023-04-09:
> Добавьте пожалуйста в https://kabinet.uk/ Челябинск

Добрый день! С мая 2022г. добавление на портал новых сайтов платное. Вы можете оплатить здесь: https://waterius.ru/sell и мы сможем отправлять показания туда. Спасибо!

dontsovcmc, 2023-04-09:
> Добрый день. Можете добавить передачу в https://rkc-m.ru/ ?

Добрый день! Можем! С мая 2022г. добавление на портал новых сайтов платное. Вы можете оплатить здесь: https://waterius.ru/sell и мы сможем отправлять показания туда. Спасибо!