dontsovcmc/waterius #199: Термодатчик

mihing, 2021-09-29:
Думаю, все сталкиваются с отключениями горячей воды. Я борюсь с этой проблемой проточным водонагревателем. Для включения его в магистраль мне необходимо переключить несколько вентилей. Если бы Ватериус имел возможность подключения термодатчика, этот процесс можно было бы автоматизировать. Например, при падении температуры ниже уставки 1, в УД передавалось бы соответствующее уведомление. И наоборот, при повышении температуры выше уставки 2, УД получал бы информацию о восстановлении подачи горячей воды.
Возможно, для УК такая информация также представляла бы интерес.

AstashovAndrey, 2021-09-30:
мне кажется это уже отдельное устройство. зачем это все городить в ватериус. Тем более что термодатчик надо в линию встраивать через переходник, снимать с трубы порочная практика(больший гистерезис, актуально только на металле). Тем более, требует постоянной работы.
Думаю самый оптимальный вариант, взять отдельно есп-шку, и к ней уже прикрутить датчик температуры, потока, ну и можно давления :) и баловаться как душе угодно :)

mihing, 2021-09-30:
Зачем – это вопрос дискуссионный. Я свои соображения высказал, ничего порочного в этом не вижу. Гистерезис не помеха, там не нужна мгновенная реакция, как не нужны и датчики потока и давления. Для вас это баловство, для других насущная необходимость. Руководствуясь вашей логикой, автоматическое снятие показаний тоже баловство. Чего бы глазами не смотреть?
Мне, допустим, было бы гораздо приятнее принять после смены горячий душ, переключив систему заранее, нежели с нецензурной лексикой выскакивая из-под ледяного душа. Наша УК не видит ничего зазорного, когда из магистрали ГВС в квартире льётся холодная вода, за которую, на минуточку, потом возьмут как за горячую.

dontsovcmc, 2022-01-19:
Крутой функционал, пока не знаю, когда добавим.

Чисто технически в версии с attiny84 можно добавить опрос термодатчика по onewire.
В настройках гистерезис (float) при котором выходить на связь.
С уставками тоже можно. Но кажется это проще на верхнем уровне сделать.
Задачу закрою, т.к. за несколько лет только 2-3 чел просили. Не первостепенно.

closed 2022-01-19 by dontsovcmc