dontsovcmc/waterius #200: Завышение показаний

svpcom, 2021-10-12:
счетчики VLF-15U-I (ХВС и ГВС), подключены как сухой контакт (не НАМУР)
Замечено, что переодически добаляются лишние импульсы. 
Общее время работы - около месяца. За первые две недели по ХВС была ошибка порядка +2 м3
После синхронизации (выставления точных значение через web ui) две недели назад: 
расход ГВС - 3.04, ХВС- 5.91
показания waterius'а : 5.914, 9.648
показания счетчиков: 5.853, 9.377
соотв. ошибки: **+2%, +4.6%**

дамп mqtt:
```
flat/water/ch0 5.914
flat/water/ch1 9.648
flat/water/delta0 90
flat/water/delta1 240
flat/water/voltage 3.060
flat/water/resets 3
flat/water/model 0
flat/water/boot 1
flat/water/good 1
flat/water/imp0 315
flat/water/imp1 615
flat/water/version 22
flat/water/version_esp 0.10.3
flat/water/voltage_low 0
flat/water/voltage_diff 0.008
flat/water/f1 10
flat/water/f0 10
flat/water/rssi -55
flat/water/waketime 5038
flat/water/setuptime 118004
flat/water/adc0 106
flat/water/adc1 106
flat/water/period_min 1440

```

это влияние помех, дребезг контактов в герконе или что-то еще?

dontsovcmc, 2021-10-16:
@svpcom спасибо за матчасть. Любопытно. Но не уверен, что это является причиной завышения показаний у вас - ведь достаточно много счетчиков valtec работают практически без погрешности. У вас прям сильно большая погрешность. А есть возможность обновить прошивку и протестировать с новым кодом adc?
Как вариант - можно digitalRead сделать и тоже проверить.

svpcom, 2021-10-16:
В прошивке один ADC переключается между двумя входами. Возможно проблема есть, когда один вход в "0", а другой "1" . Холодная вода как раз на втором входе и возможно ADC не хватало времени перезарядить конденсатор с 0 до 1 и код считал это завершением импульса. Но это только мое преположение.

Я обновил прошивку - за пару дней пока лишних импульсов не набежало. Подожду неделю - две и будет видно - помогло или нет

dontsovcmc, 2021-10-16:
@svpcom "Возможно проблема есть, когда один вход в "0", а другой "1" " - Любопытная гипотеза! Пошёл за попкорном, что получится. Но картина у всех пользователей схожая...

dontsovcmc, 2021-11-09:
@svpcom как успехи тестирования?

svpcom, 2021-11-10:
Провел испытания:
Новые показания счетчиков:
хвс: 11.477, гвс:  7.405
waterius:
хвс: 11.777, гвс: 7.484

погрешность хвс: (11.777 - 11.477) / (11.477 - 9.377) = 14.2 %
погрешность гвс: (7.484 - 7.405) / (7.484 - 5.914) = 5%

судя по всему, что показания всегда завышаются, то где-то не компенсируется дребезг контактов. Также так как гвс опрашивается первым (ch0), то возможно взаимное влияние их друг на друга.  Я думаю, что надо поправить процедуру проверки импульса - добавить условие, что начало импульса это не любое замыкание контактов, а только то, что длится более некоторого количества времени (как это сделано с размыканием)

svpcom, 2021-11-12:
@dontsovcmc А в чем смысл замены attiny85 на attiny45? Это привело к тому, что сейчас осталось всего свободно 44 байта для нового кода.

dontsovcmc, 2021-11-12:
> @dontsovcmc А в чем смысл замены attiny85 на attiny45? Это привело к тому, что сейчас осталось всего свободно 44 байта для нового кода.

В мире кризис микроконтроллеров, поэтому чем больше выбор, тем проще найти их у поставщиков =).

svpcom, 2021-11-19:
@dontsovcmc Поправил pull-request. Теперь вроде точно все считает

dontsovcmc, 2022-01-19:
@svpcom обнаружил 2-х пользователей у кого 1л/имп и пропускают показания.
при 1л/имп очень быстро проходит импульс.
#define TRIES_OPEN  24 - не подойдёт =(.

ощущение, что нужно на прерывания переходить + проверка дребезга. что думаете?

svpcom, 2022-01-19:
@dontsovcmc Мне кажется тут надо ставить внешнюю подтяжку и включать ее полевым транзистором по команде с МК (когда он просывается). Но тут потребуется другой МК, так как у attiny ноги уже закончились. Без нее namur помехоустойчиво не получится считать. Ну и придется сделать в прошивке выбор типа счетчика: namur/dry contacts и 1/10 литров на импульс для выбора допустимых границ напряжений и задержек. Так как без этого точно считать не получится.  Вообще я бы перешел с attiny на stm32f0/f1 - там памяти сильно больше, сильно лучше переферия и его можно удаленно препрошивать через загрузчик.

dontsovcmc, 2022-01-19:
@svpcom namur/dry тоже можно автоматически + ручной ввод сделать, по уровню во время настройки...  наверно мне стоит собрать статистику по корректности показаний у кого Намур и тогда можно сделать вывод о необходимости в этом. А помехоустойчивость хардварно не улучшить? (кроме внешней подтяжки)

svpcom, 2022-01-19:
@dontsovcmc А как ее улучшить, если при такой слабой подтяжке помеха получается больше сигнала? Я когда осцилограф подключит там сплошной шум был. В многоэтажных железзобетонных домах с электроплитами похоже вдоль стояка водопровода ходят приличные блуждающие токи.   А разве есть сейчас на рынке счетчики с OneWire?

dontsovcmc, 2022-01-19:
@svpcom 
Подтяжка, кстати 30к где-то. Геркон даёт 110 при замыкании (а там 3к3 резистор стоит).

OneWire между attiny и esp.

А шумы будут при герконе или намуре?

neitri, 2022-02-12:
Мне кажется алгоритм "анти дребезга" несколько не правильный.
Когда напряжение падает ниже уровня LIMIT_CLOSED или LIMIT_NAMUR_CLOSED на время одного цикла счета. Инициируется таймер значением 3. 
Возрастании напряжения на входе. В течении 3 циклов считает таймер и по истечению времени увеличивает счетчик.
На мой взгляд отсутствует гистерезис. Меньше 170 контакт замкнут, А больше 170 уже инициируем счетчик и ждем подсчета импульсов.
детектор разомкнутого состояния должен быть несколько иной.
Разомкнут если больше LIMIT_OPEN ( 800) иначе если больше LIMIT_NAMUR_OPEN (допустим 400) разомкнут намур.
Сам же подсчет импульсов можно отвязать от статусов  и если напряжение меньше LIMIT_NAMUR_CLOSED (200) считаем таймер анти дребезга. А если напряжение больше LIMIT_NAMUR_OPEN (400) взводим таймер.

dontsovcmc, 2022-02-20:
> @dontsovcmc По даташиту attiny сопротивление внутреннего подтягивающего резистора нахожится в пределах 20 - 50 кОм, в то время как ADC требует выход с сопротивлением не более 10 кОм https://electronics.stackexchange.com/questions/107741/required-output-impedance-for-adc-input.
> 
> https://github.com/dontsovcmc/waterius/blob/1816e56ca21b2f08f4d675c4b5dc35241a237dc2/Attiny85/src/counter.h#L53
> 
> 
> Мне кажется, что тут надо после включения подтягивающего резистора поставить некоторую задержку и потом в цикле проверят значение adc, пока оно не стабилизируется

Занялся адпейдом прошивки и внимательнее изучил вопрос. Не хочется мне "с запасом" добавлять 1мс, т.к. там потребление 800мкА, что дает в сутки эквивалент 350мА*1сек.
По ссылке: 
With R = 100 kΩ and C = 14 pF, the RC time constant (TC) is 1.4 μsec.
У нас 40кОм - 0.5 мкс. Т.е. 1000мкс выглядят лишними. 
У вас есть возможность убрать эту задержку и проверить точность подсчета?

svpcom, 2022-02-22:
Тут надо с осцилографом и генератором сигналов отлаживать. В теории analogRead() отрабатывает за 0.1ms.

svpcom, 2022-02-22:
Возможности поменять прошивку сейчас у меня нет

svpcom, 2022-02-22:
Мне кажется что фундаментальная проблема тут другая:
```
 // Возвращаем текущее состояние входа
    enum CounterState_e value2state(uint16_t value)
    {
        if (value < LIMIT_CLOSED) {
            return CounterState_e::CLOSE;
        } else if (value < LIMIT_NAMUR_CLOSED) {
            return CounterState_e::NAMUR_CLOSE;
        } else if (value < LIMIT_NAMUR_OPEN) {
            return CounterState_e::NAMUR_OPEN;
        } else {
            return CounterState_e::OPEN;
        }
    }
```
данная функция не имеет запрещенных состояний. То есть любое напряжение на входе конвертируется в состояние выхода. Поидее ее надо переписать так:
-  U < NAMUR_CLOSE -- closed
-  U > NAMUR_OPEN  -- open
- else undefined

так как иначе любая помеха начинает отсчитывать таймер

> Мне кажется алгоритм "анти дребезга" несколько не правильный. Когда напряжение падает ниже уровня LIMIT_CLOSED или LIMIT_NAMUR_CLOSED на время одного цикла счета. Инициируется таймер значением 3. [...]

wrewolf, 2022-04-15:
Подпишусь, у меня такая же проблема

dontsovcmc, 2022-11-16:
> Подпишусь, у меня такая же проблема

А какой марки у вас счетчики? Пока таких проблем больше не встречал. Но встречал счетчики с "двойными" импульсами — может как-то магнит там дефектно установлен.

dontsovcmc, 2023-04-11:
Закрою задачу. Это какой-то редкий кейс.
Да, в ветке universal код теперь есть возможность по изменению уровня считать импульсы. Может в вашем случае это подойдет?

closed 2023-04-11 by dontsovcmc

wrewolf, 2023-04-11:
Я как раз 2 недели как обновил прошивку, наблюдаю

wrewolf, 2023-04-11:
У меня 45, изначально прошивка была 0.10.5