dontsovcmc/waterius #213: Понизить разрядность АЦП

neitri, 2022-02-09:
Есть предложение снизить разрядность АЦП.
У микроконтроллера есть бит ADLAR. При значении в нем 0, значение АЦП записывается 2 старших разряда в ADCH и 8 младших в ADCL. И получаем вполне понятное значение на выходе.
При ADLAR=1 значение записывается 8 старших разрядов (из 10) в ADCH а 2 младших разряда записываются в 2 старших разряда ADCL. Т.е. значение АЦП записывается сдвинутым в лево на 6 разрядов. 
Таким образом можно намеренно потерять в точности 2 разряда и оперировать исключительно старшей частью ADCH без каких либо математических расчетов. Все константы необходимо будет уменьшить в 4 раза. 
Преимущества: сокращение кода за счет оперирования переменными меньшего размера. Все сравнения, присваивания, обращение к памяти будут занимать по 1 слово и 1 такт. Для 16 битных значений будет по 2 слова и 2 такта.
Есть обратная совместимость. Если не считать уменьшенного значения в 4 раза.
Экономия получается около 40 байт флеш и 2 байта рам.

dontsovcmc, 2022-02-20:
А для чего экономить размер кода? 
Из оптимизаций - сейчас уберем чтение напряжения. Также я не уверен, что нужен кольцевой буфер - надо смотреть, насколько устройства перезагружаются по БД.

neitri, 2022-02-21:
Можно уменьшить время исполнения.
Читает с ацп 1 байт (2байта). Сохроняет в рам 1 байт (2). Загружает константу для сравнения 1 байт (2). Читает с рам 1 байт (2). Сравнивает 1 байт (2). Меньше передавать в есп. А по i2c это 23 байта за 22 мс передается если я не ошибаюсь.
В итоге можно сократить время подсчета. 
Другой момент постепенно уйти от arduino.

dontsovcmc, 2022-02-21:
Насколько я понимаю, экономия операция не даст большого экономия потребления, потому что само чтение из аналогового входа достаточно долгое:

[image]

Пришлите свою версию прошивки, я измерю потребление.
Оптимизировать i2c тоже нет смысла, потому подключение к роутеру занимает секунд 5 и в этот момент ESP жрёт в среднем 80 мА, поэтому экономия 22мс не даст прироста =(. Эти оптимизации для радиоканала хороши.

dontsovcmc, 2025-06-08:
Как идея - ок. Но оптимизация кажется избыточной. Закрою задачу, чтобы не болталась.

closed 2025-06-08 by dontsovcmc