dontsovcmc/waterius #222: Время подключения к сети.

neitri, 2022-02-28:
Обратил внимание, что время подключения зависит от используемого канала. У меня был 13 канал динамический адрес, время работы около 6-7 секунд. Со статическим адресом около 5. 
Поменял канал на 1 и подключение произошло буквально мгновенно. 1-2 раза напряжение успевает измерить.
Но тут ечть вопросы и к себе.
Я пробывал указывать и канал и мак адрес точки доступа. Но даже если я указываю при подключении 13 канал, время подключения увеличивается. При подключении esp сканирует все каналы пока не найдет свою сеть. Почему при указании канала все равно их сканирует мне не понятно. 

Давайте попытаемся собрать статистику у кого какой канал и сколько время подключается.

Tls соединение 2-4 секунды занимает. А то и все 6. Но там вопросы к качеству интернета.

dontsovcmc, 2022-02-28:
Я видел большие темы по поводу каналов Wi-Fi, но не вдавался в подробности.. А мы можем сохранять канал в памяти и подключаться к конкретному?

dontsovcmc, 2022-02-28:
[тема на гитхабе про время подключения](https://github.com/espressif/arduino-esp32/issues/1675#issuecomment-850994388)

Тут документ просто про частоты и легитимность в странах: https://www.espressif.com/sites/default/files/documentation/esp8266_wi-fi_channel_selection_guidelines_en.pdf

neitri, 2022-03-01:
Эта тема мне уже попадалась. Но результат пока отрицательный.
Указание мак адреса сети вроде ускоряет процес подключения. Но сканирование каналов никуда не девается. С 1 каналом подключается быстрее. 
На данный момент получается что для сокращения времени работы необходимо установить статический ip адресс и использовать по возможности наименьший канал.

По прописыванию мак адреса точки доступа время подключения может и сокращается, но время с отправкой занимает примерно тоже самое.

dontsovcmc, 2022-12-18:
@neitri я кстати подумал, что хорошо бы для эксперимента уменьшить версию прошивки фреймворка. Потому что у кого-то типа esphome используется одна из первых версий прошивки как stable.
эксперимент может быть: очистка памяти  (чтобы настройки вайфая точно стерлись), прошивка версией Х с захардкоженными настройками wi-fi, замер времени подключения.

vzagorovskiy, 2023-02-15:
fixed https://github.com/dontsovcmc/waterius/commit/76eb2bf4c77cbecfdc889b3541fd2145284fe187

dontsovcmc, 2023-04-29:
да, 0.11.0 подключается за 2.5 сек. соберем статистику. посмотрим еще

closed 2023-04-29 by dontsovcmc