dontsovcmc/waterius #231: Синхронизация времени убегает вперед (раз в час)

dontsovcmc, 2022-09-06:
У пользователя dmit.kam период раз в час. Кажется Ватериус всё время на минуту позже выходит на связь.

neitri, 2022-09-08:
Что то не совсем понятно, заголовок как то не соответствует содержанию.
Я могу предположить, что отправка показаний 1 раз в час с учётом постройки периода может давать такой эффект.
Тинька считает 60 "минутных" периода. Выходит на связь. Рассчитывает адаптацию по фактического времени. Там 61 с копейками минута. Получается что нужно выходить на связь 58.9 минут но дробных нет и период будет 58 "минутных" периодов. А минута у ватериуса 59 секунд как минимум.
В любом случае как то сильно точно подсчитать период мне кажется не получится.
Второй момент. Если мне не изменяет память, во время отправки показаний время ватериус не считает. Т.е. досчитал ватериус до 0, включает есп отправляет, рассчитывает новый период отправки, отправляет данные,  передаёт период в тиньку, и командует спать. Только после команды спать начинается отсчёт времени. Как минимум при таком раскладе каждая отправка должна сдвигать время отправки на время отправки.
Можно конечно замарочится. Передавать ещё в тиньку количество пробуждения в последнюю минуту. Тогда можно будет точнее считать периоды. Но в тоже время необходимо постоянно считать время. Сторожевой таймер должен постоянно работать. И считать время.

dontsovcmc, 2023-02-16:
возможно исправили в 0.11.0

closed 2023-04-11 by dontsovcmc

wrewolf, 2023-04-11:
Не до конца, теперь бежит по секундам
17:35:03
18:35:06
19:35:10
20:35:14
21:35:22
22:35:26
23:35:30
Не было изменений
01:35:36

// тоже ночь без изменений

08:36:01
09:36:05
10:36:09

dontsovcmc, 2023-04-11:
> Не до конца, теперь бежит по секундам 17:35:03 18:35:06 19:35:10 20:35:14 21:35:22 22:35:26 23:35:30 Не было изменений 01:35:36

Кайф, это корректно - погрешность кругами ходить будет от 0 до 60 сек, т.к. цикл по минутам сделан.

wrewolf, 2023-04-11:
А если станет 37:01, значит не корректно? Я понаблюдаю

dontsovcmc, 2023-04-11:
> А если станет 37:01, значит не корректно? Я понаблюдаю

я думаю до 37:05 корректно и дальше сбросится в 36:05, но это гипотеза

wrewolf, 2023-04-12:
Не, идёт вперёд, последняя 12:37:46