dontsovcmc/waterius #233: Поддержка "электронных" импульсов

dontsovcmc, 2022-09-28:
Хочется поддержать максимальное кол-во оборудования. Кажется характер электронных импульсов разный. Здесь будет ссылка на техпаспорта и описание подключений.

## Счетчики газа
### СГБМ-1,6
В конструкции электронного блока применяется оптрон типа PC357N4T
(или аналогичный), позволяющий применять дистанционный съем показаний, который выдает импульс во внешнюю сеть учета на каждые 10 литров прошедшего
газа со следующими параметрами:
- ток внешней цепи оптрона не более 3 мА;
- напряжение на внешней цепи оптрона не более 35 В;
- длительность импульса не менее 15 мс.

## Счетчики электроэнергии 
### Меркурий 230

Номинальное напряжение, подаваемое на импульсный выход равно 12 В (предельное - 24 В).
Номинальный ток импульсного выхода - 10 мА (предельный - 30 мА).
[sgbm-16-rukovodstvo.pdf](https://github.com/dontsovcmc/waterius/files/9662268/sgbm-16-rukovodstvo.pdf)
[merkuriy-230-avlg-411152-021-re-izm-3-2018-04-01.pdf](https://github.com/dontsovcmc/waterius/files/9662271/merkuriy-230-avlg-411152-021-re-izm-3-2018-04-01.pdf)

StrannikP, 2022-10-01:
Простой переход в deep sleep не решит всех проблем. Дело в том, что контакт должен обтекаться током, иначе как мы заметим, что он замкнут или разомкнут. Пока условно, примем величину тока в 1мА. При более низких значениях любое включение элктроприбора, например фена, будет давать серию помех на которые будет реагировать МК. Значит надо ставить мультивибратор, который будет подавать импульсы в 1мА с заданной периодичностью и скважностью.

dontsovcmc, 2022-10-02:
> любое включение элктроприбора, например фена, будет давать серию помех 

Нам микропотребление нужно при питании от батареек. В этом же случае фен не помешает?
Если питание от сети, то можно вообще не спать.

StrannikP, 2022-10-03:
Вы путаете электрические помехи по питанию и радиопомехи. Если телефон лежит рядом с колонками, вы слышите характерный звук из колонок, когда  вам на него звонят. Это и есть радиопомехи, ведь телфон не связан электически с колонками. Фен, из-за наличия щёточного мотора, также создает радиопомехи. Для защиты от таких помех существует много способов, в том числе и увеличение тока больше заданного порога. Так для длинных цепей например Ethernet или RS485 ток может быть 100 мА. Для более коротких например LVDS 3.5 мА. Это не такой простой вопрос, как кажется на первый взгляд. И его решение должно иметь комплексный характер.

dontsovcmc, 2023-03-21:
Вполне возможно, что расстояние между импульсами будет короче... Но в целом, вы правы. 
Мы сейчас тестируем прошивку для импульсов. У вас будет возможность проверить её?

dontsovcmc, 2023-10-28:
support Берил

dmitrypov, 2023-12-05:
Удалось ли подружиться с ИПУ:

- Decast МАРС СТК (отопление),
- Mercury 200.02 (электроэнергия)?

Планируются ли другие версии УСПД, например с nb-iot, в защищенном корпусе ip65?