dontsovcmc/waterius #234: Добавить настройку параметра retained для mqtt параметров

Spirituss, 2022-10-04:
Просьба добавить настройку параметра retained для mqtt параметров для возможности автоматического обновления значений в УД.

dontsovcmc, 2022-10-04:
Добрый день! А в каких случаях требуется retained=false? Сейчас он всегда true.

Spirituss, 2022-10-04:
> Добрый день! А в каких случаях требуется retained=false? Сейчас он всегда true.

Не знаю по какой причине, но у меня сейчас все сообщения в топике с false

dontsovcmc, 2022-10-04:
> > Добрый день! А в каких случаях требуется retained=false? Сейчас он всегда true.
> 
> Не знаю по какой причине, но у меня сейчас все сообщения в топике с false

Задайте вопрос в телеграм чате waterius_forum ... В коде true: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/sender_mqtt.h#L40

closed 2022-10-09 by Spirituss