dontsovcmc/waterius #235: Отображать 3 знака после запятой в личном кабинете

Spirituss, 2022-10-09:
Сейчас нет возможности в режиме реального времени делать сверку показаний в MQTT и в личном кабинете. В MQTT показания с 3 знаками, в личном кабинете - округление. Приходится долго ждать, чтобы понять, есть ли расхождение.

dontsovcmc, 2022-10-09:
В MQTT и личный кабинет отправляются одинаковые числа, зачем их сравнивать? =)

neitri, 2022-10-11:
А можно уточнить цену импульса на ваших счетчиках?
Предполагаю что 10 литров, 0.01м³.

Spirituss, 2022-10-11:
> А можно уточнить цену импульса на ваших счетчиках? Предполагаю что 10 литров, 0.01м³.

10 литров. Но вопрос не в этом, а в том, что если округляем - то в обоих местах, если нет - то так же, в обоих показываем как есть. В ватериус начальные показания вносил 3 знака, там так 3 знака и сохраняются.

neitri, 2022-10-11:
Для отображения 3го знака после запятой необходимы счетчики с весом импульса в 1 литр на импульс. При цене в 10 литров, 3й знак будет всегда 0.

dontsovcmc, 2022-10-17:
> > А можно уточнить цену импульса на ваших счетчиках? Предполагаю что 10 литров, 0.01м³.
> 
> 10 литров. Но вопрос не в этом, а в том, что если округляем - то в обоих местах, если нет - то так же, в обоих показываем как есть. В ватериус начальные показания вносил 3 знака, там так 3 знака и сохраняются.

Идея в том, что личный кабинет - он для обычного пользователя. Нет потребности в 3х знаках после запятой.
А MQTT вы вольны настраивать, как вам хочется - с 2, 3 знаками. Поэтому смешивать не имеет смысла. Предлагаю закрыть задачу =)

Spirituss, 2022-10-19:
> Идея в том, что личный кабинет - он для обычного пользователя. Нет потребности в 3х знаках после запятой. А MQTT вы вольны настраивать, как вам хочется - с 2, 3 знаками. Поэтому смешивать не имеет [...]

Решил задачу другим способом - в ха сенсор сделал с 2 знаками + в ватериус в качестве стартовых данных указал тоже 2 знака.

closed 2022-10-19 by Spirituss