dontsovcmc/waterius #242: Сбой периода пробуждения

dontsovcmc, 2022-12-21:
[image]

Стоит 60 минут, потом почему то раз в 15 мин начал отправлять.

Не очень понятно, как отлаживать.

Ещё любопытно, как на расчёты повлияет не получение точного времени по NTP...

neitri, 2022-12-23:
Без синхронизации времени не имеет смысла изменение #209 
Пересчет периода необходимо делать только при работающей синхронизации времени. Само просто наверно повторно попытаться синхронизировать время. 
                
```
//Перерасчет времени пробуждения
if (mode == TRANSMIT_MODE) { 
```
изменить на 
```
//Перерасчет времени пробуждения
if (mode == TRANSMIT_MODE && setClock()) {
```

dontsovcmc, 2022-12-23:
Да, подумал про это...

Ещё вопрос:
1. Что вернёт time(nullptr), если не будет синхронизировано время?
2. Может сделать, если время не синхронизировано, то не выполняем уточнение периода, а оставляем константу?
3. При каких обстоятельствах t1 <= 1?

```
time_t now = time(nullptr);
  time_t t1 = (now - sett.last_send) / 60;
  if (t1 > 1 && data.version >= 24)
                    
```

neitri, 2022-12-23:
Если период отправки минуту поставить, то может получится t1=0 а это не хорошо будет.

dontsovcmc, 2023-01-03:
Да, в прошивке Tasmota определяется константа, если время ниже ее, то значит оно не получено. 
Надо тоже сделать такую константу.

https://github.com/arendst/Tasmota/search?q=START_VALID_TIME

const uint32_t START_VALID_TIME = 1451602800;  // Time is synced and after 2016-01-01

Сейчас должно работать так: если NTP не получено, то мы опять засыпаем. Так ли это?
Если не так, то надо сделать чтобы засыпали.

neitri, 2023-01-04:
На мой взгляд, нужно синхронизацию времени выносить куда то выше. И сохранять флаг что время синхронизировано.
Подстройка времени работать будет только с синхронизацией времени обоих передач по расписанию. Т.е. если время не синхронизировано, то при следующей отправки подстроить время не удастся, т.к. время отправки показаний будет не корректное.
Как вариант пользовательский сервер времени использовать.

dontsovcmc, 2023-01-12:
Пользователь подтвердил сбой периода
[image]

vzagorovskiy, 2023-02-15:
fixed #249

closed 2023-04-11 by dontsovcmc