dontsovcmc/waterius #245: MQTT

vzagorovskiy, 2023-01-06:
- [ ] Добавить возможность выбора типа публикации данных в MQTT, одним топиком с json или 30 отдельными топиками
- [ ] При публикации данных вне зависимости от типа публикации данные брать из UserClass, чтобы не разнились названия полей/топики и их содержимое
- [ ] Переделать публикацию в отдельные топики, используя итерацию по JSON документу,  пример ниже
```cpp
for (JsonPair keyValue : documentRoot) {
    Serial.println(keyValue.key().c_str())
    Serial.println(keyValue.value().as<char*>());
  }
```
см. инфо [JsonObject::begin() / JsonObject::end()](https://arduinojson.org/v6/api/jsonobject/begin_end/) 
- [ ] Переименовать UserClass в более точно описывающих характер модуля
- [ ]  Добавить опцию добавлять ли discovery топик для HomeAssistant в MQTT
- [ ] Указывать discovery_prefix для discovery топик для HomeAssistant в MQTT _(Под вопросом, отложить на потом)_ 

@dontsovcmc

dontsovcmc, 2023-01-06:
- Cоздать переменную в конфигурации ESP bool conf.ha_discovery  # homeassistant discovery
- Добавить параметр в веб-морду чекбокс "homeassistant discovery" который влияет на conf.ha_discovery. id=ha.
 
- При публикации данных брать json из UserClass, чтобы не разнились названия полей/топики и их содержимое.
- Переделать публикацию в отдельные топики, используя итерацию по JSON документу, пример ниже
for (JsonPair keyValue : documentRoot) {
    Serial.println(keyValue.key().c_str())
    Serial.println(keyValue.value().as<char*>());
  }
см. инфо [JsonObject::begin() / JsonObject::end()](https://arduinojson.org/v6/api/jsonobject/begin_end/)

- Переименовать UserClass в более точно описывающих характер модуля. Пример: http_sender.cpp

dontsovcmc, 2023-01-06:
Под вопросом, т.к. тогда нужно будет всегда синхронизировать время. А доступа к NTP серверу из локальной сети умного дома может не быть... т.е. нужно тогда делать параметр "NTP сервер". 
- Добавить в данные timestamp или текущее время, ноподобии как в Zigbee2MQTT (last_seen)

Давайте в след. шаге?

closed 2023-02-15 by vzagorovskiy