dontsovcmc/waterius #246: Поиск причины ошибки подключения к некоторым роутерам

dontsovcmc, 2023-02-02:
Для того, чтобы выяснить причину ошибки подключения к роутерам (https://github.com/dontsovcmc/waterius/issues/229, https://github.com/dontsovcmc/waterius/issues/131) предлагаю:

При наличии «плохого» роутера:

1. Залить пример обычного подключения к WiFi последней версии arduinoframework.
2. Залить пример из WiFiManager последней версии (master ветку https://github.com/tzapu/WiFiManager/tree/master) и проверить, подключиться ли к роутеру.
3. Залить https://github.com/arendst/Tasmota последней версии и проверить, подключиться ли к роутеру.

Важно выяснить при каких настройках роутера возникает ошибка, поэтому сначала стоит перебрать эти варианты, а затем уже перенастраивать роутер.

vzagorovskiy, 2023-02-15:
partially fixed #247 #252

dontsovcmc, 2023-07-05:
не подключается:
[image]

dontsovcmc, 2023-08-07:
Нашёл в коде ESP описание точки доступа. Все поля требуется вывести, чтобы видеть отличия.

```
struct bss_info {
    STAILQ_ENTRY(bss_info)     next;

    uint8 bssid[6];
    uint8 ssid[32];
    uint8 ssid_len;
    uint8 channel;
    sint8 rssi;
    AUTH_MODE authmode;
    uint8 is_hidden;
    sint16 freq_offset;
    sint16 freqcal_val;
    uint8 *esp_mesh_ie;
    uint8 simple_pair;
    CIPHER_TYPE pairwise_cipher;
    CIPHER_TYPE group_cipher;
    uint32_t phy_11b:1;
    uint32_t phy_11g:1;
    uint32_t phy_11n:1;
    uint32_t wps:1;
    uint32_t reserved:28;
};
```

Fortunatto, 2023-08-27:
Не подключается к роутеру в принципе.
Роутер - Mikrotik hap ac^2 (RBD52G-5HacD2HnD-TC) - версия прошивки 7.11.
Waterius - esp v. 0.11.7.

Менял настройки роутера: стандарты связи (b/g/n и их комбинации), ширину каналов, страну использования (влияет на используемые частоты, как я понимаю) - ничего не помогло. Кстати, при выборе страны "rissia 4", esp Ватериуса мою сеть не видела в принципе.

В качестве теста залил на esp: [Tasmota 13.1.0 by Theo Arends](https://bit.ly/tasmota) - все подключилось и работает.

dontsovcmc, 2023-08-27:
> Не подключается к роутеру в принципе. Роутер - Mikrotik hap ac^2 (RBD52G-5HacD2HnD-TC) - версия прошивки 7.11. Waterius - esp v. 0.11.7.
> 
> Менял настройки роутера: стандарты связи (b/g/n и их комбинации), ширину каналов, страну использования (влияет на используемые частоты, как я понимаю) - ничего не помогло. Кстати, при выборе страны "russia 4", esp Ватериуса мою сеть не видела в принципе.
> 
> В качестве теста залил на esp: [Tasmota 13.1.0 by Theo Arends](https://bit.ly/tasmota) - все подключилось и работает.

Ооо низкий поклон за такое подробное исследование! 

Кажется код подключения к точке доступа тут:
https://github.com/arendst/Tasmota/blob/23d6c0f5d5f2882a3c744669cdcbfee832f5ab52/tasmota/tasmota_support/support_wifi.ino#L203

У WifiManager вот (наша ветка):
https://github.com/dontsovcmc/WiFiManager/blob/waterius_release_112/WiFiManager.cpp#L1143

dontsovcmc, 2023-08-27:
> Не подключается к роутеру в принципе. Роутер - Mikrotik hap ac^2 (RBD52G-5HacD2HnD-TC) - версия прошивки 7.11. Waterius - esp v. 0.11.7.

А точка доступа содержит пробелы? Если да, то точно из-за них. В веб морде вручную введите пожалуйста пробел.

vzagorovskiy, 2023-08-27:
wAP ac, cAp ac, hAp ac2 у всех 7.11  работает все норм. Проблема в конфигурации.

Fortunatto, 2023-09-03:
> А точка доступа содержит пробелы? Если да, то точно из-за них. В веб морде вручную введите пожалуйста пробел.
Имя точки доступа без пробелов, содержит заглавные и строчней буквы, цифры и знак "-". До этого прошивку своих Ватериусов обновлял в 2022 году - переход прошел безболезненно.

> Ещё вопрос: а до обновления была прошивка уже? стоит очистить память перед новой прошивкой..
Чистил память заливкой пустой прошивки.

> hAp ac2 у всех 7.11 работает все норм. Проблема в конфигурации.
Имеете в виду конфигурацию роутера? Tasmota же подключилась...

dontsovcmc, 2023-09-03:
Хм не знаю, что предположить тогда.
Я увидел, что wifimanager посадил ошибку с пробелом в имени сети. Может конечно они с тире тоже сделали такое :(. Удобно проверить: вручную введя имя точки доступа в строку под списком вайфаев?

0.11.8 исправлен только пробел, можно на 0.11.7 проверить. Спасибо!
А вам удобно снять лог с вывода LOG у ESP? Очень пригодится

Fortunatto, 2023-09-09:
> wAP ac, cAp ac, hAp ac2 у всех 7.11 работает все норм. Проблема в конфигурации.

Возможно Вы правы - попробовал подключиться к виртуальной ТД на телефоне - соединение прошло. Можете поделиться своей конфигурацией?

dontsovcmc, 2023-09-09:
> > wAP ac, cAp ac, hAp ac2 у всех 7.11 работает все норм. Проблема в конфигурации.
> 
> Возможно Вы правы - попробовал подключиться к виртуальной ТД на телефоне - соединение прошло. Можете поделиться своей конфигурацией?

А вы тоже можете поделиться своей?

Ещё недавно писали, что не подключается к only n. А если bgn, то норм

Fortunatto, 2023-09-09:
> А вы тоже можете поделиться своей?
>
> Ещё недавно писали, что не подключается к only n. А если bgn, то норм

Подтверждаю. Перешел в b/g/n, ширину канала перевел в 20 МГц - удалось подключиться. Видимо это в совокупности и сработало. В прошлый раз менял на b/g/n, но это не помогло.

dontsovcmc, 2023-10-31:
В версии 1.0.0 подключаемся по другому.
Нужно проверить её с проблемными роутерами!

closed 2023-12-25 by dontsovcmc