dontsovcmc/waterius #254: Обновить версию WiFiManager

dontsovcmc, 2023-02-18:
WiFiManager выложен в декабре 2022 г.
Наш 2020г и можно обновить. Возможно это поможет с проблемными роутерами, т.к. первичное подключение именно WiFiManager делает.

dontsovcmc, 2023-02-23:
> Hi, Have you included local Blynk server extra settings in this release?

Haven’t add..
But maybe you help us?
What parameters we should add to settings page?
host: string
port: integer
?

Stas15, 2023-02-23:
Honestly I’m not good in programming, I’m just end user. But I believe if you could add host,IP and port for local Blynk server in WiFi manager it will be a great job which could solve issues with this integration. Thank you in advance.

dontsovcmc, 2023-02-23:
> Just as mine pain, now I cannot connect waterius with mobile app. It connects, but dosnt send data to App, and sends only to your cloud. Im happy with it, but I liked it in my phone. I have no problems with my other projects, but to be honest, they much simpler then yours.

Do you have you own Blynk server?

Stas15, 2023-02-23:
Yes, but it is not finished, there are few problems to solve

Stas15, 2023-02-26:
Solved :)

closed 2023-04-29 by dontsovcmc