dontsovcmc/waterius #277: Вебинтерфейс настройки: замечания, пожелания

dontsovcmc, 2023-04-29:


dontsovcmc, 2023-04-29:
Известные мне проблемы:
1. При повторной настройке надпись "не подключено" сильно смущает пользователей — они опять выливают воду. 
Делать это не требуется.
Решение: в JS скрипте посмотреть - если указан вес импульса, то не показывать табличку.
При первой настройке в весе импульса указано "авто" и "как у холодной" - это признак, что юзер первый раз настраивает.

2. Изменение WIFI сети не удаляёт звёздочки в пароле. 
Решение: очищать поле пароль, если пользователь сменил WIFI сеть.

3. Не понятно какая версия attiny, ESP.
При настройке, если проблемы, то хорошо бы это знать.
Решение: Пожелание добавить номера версий ESP и attiny на первую или страницу настроек (чтобы когда делали скриншот на нём было это видно)

4. На странице после Сохранить не отображается статус подключения Ватериуса к роутеру. По хорошему нужно отображать: плохой пароль и др. ошибки.

embisda, 2023-04-29:
[image]
Звездочки в поле пароль сбивают с толку. Непонятно что туда нужно вбивать пароль и поле на самом деле незаполнено

dontsovcmc, 2023-05-20:
Если MQTT топик пустой, добавлять косую черту принудительно)

dontsovcmc, 2023-10-31:
В 1.0.0 всё сделано. Ура, товарищи! =)

closed 2023-10-31 by dontsovcmc