dontsovcmc/waterius #278: Второй раз фиксирую зависание устройства

wrewolf, 2023-05-02:
Вот так выглядит в хом асистант, в чуство приходит нажатием кнопки для ручной отправки.
[image]

Завис около 18:00 27 апреля, только сейчас заметил тыкнул кнопочку и он послушно и без проблем отправил показания.

Версия программного обеспечения: 0.11.3.25
Версия аппаратного обеспечения: 1.0.0

dontsovcmc, 2023-05-02:
хмм. а у вас на наш сайт отправляются данные? если да, пришлите мне адрес почты, пожалуйста.
А можете добавить в HA график параметра resets и boot?

wrewolf, 2023-09-26:
Больше не повторялось. Создам новую проблему если будет повторяться

closed 2023-09-26 by wrewolf