dontsovcmc/waterius #309: Госуслуги дом запустили по всей стране.

WeslyG, 2024-01-21:
День добрый, госуслуги дом запустили на всей территории страны. 
https://blog.domclick.ru/novosti/post/prilozhenie-gosuslugi-dom-zapustili-po-vsej-rossii

Я думаю теперь нет смысла интегрировать тыщи разных сервисов, когда можно интегрировать один, с понятным аутхом, и всем прочим. 

Рассматриваете ли вы для себя такую задачу?

Я раскрою чуть подробнее, у меня прямые расчеты с компаниями которые предоставляют услуги, итого отправлять данные мне нужно в 4 места, что очень не удобно, даже имея на руках все данные, а в госуслугах я могу это сделать с единого места

dontsovcmc, 2024-01-29:
Спасибо за информацию! 
По нашей информации у ГИС ЖКХ пока нет сущности "умный счётчик" =(. В 2024-м планируют реализовать и мы будем говорить про интеграцию.

А в ваших 4-х компаниях есть интернет-сайты или они в ГИС ЖКХ принимают данные?

WeslyG, 2024-02-08:
У каждой компании свой сайт, свой личный кабинет, свой метод приема данных (разумеется мало у кого есть апи). И буквально одна есть в ватериусе.