dontsovcmc/waterius #314: Добавить в discovery HA информацию о настройке полей

dontsovcmc, 2024-02-20:
HA знает только о том , что можно период поменять. А нужно еще добавить информацию про другие поля.

Anat0l, 2024-03-04:
> HA знает только о том , что можно период поменять. А нужно еще добавить информацию про другие поля.

О каких полях идет речь? Можно подробней раскрыть мысль?

dontsovcmc, 2024-03-05:
HA получает через discovery запрос информацию о параметрах устройства. В том числе информацию read / write. Сейчас только у периода есть write. А я добавил изменение и других полей, но не поменял информацию в discovery json

Anat0l, 2024-03-06:
> HA получает через discovery запрос информацию о параметрах устройства. В том числе информацию read / write. Сейчас только у периода есть write. А я добавил изменение и других полей, но не поменял информацию в discovery json

еще раз спрошу, каких других полей?

dontsovcmc, 2024-03-06:
> > HA получает через discovery запрос информацию о параметрах устройства. В том числе информацию read / write. Сейчас только у периода есть write. А я добавил изменение и других полей, но не поменял информацию в discovery json
> 
> еще раз спрошу, каких других полей?

вот здесь  в коде они есть: 
https://github.com/dontsovcmc/waterius/blob/fdfecc5f93b94369dfadd142dccf43055aac3f38/ESP8266/src/ha/subscribe.cpp#L51

Anat0l, 2024-03-06:
в  ESP8266/src/ha/resources.h
Поменять тип на необходимых записях `sensor_type` с `s_sensor` на `s_number` 
Возможно понадобится дополнительная настройка для значений типа float

dontsovcmc, 2024-03-26:
@Anat0l спасибо! 

Версия 1.1.2

closed 2024-03-26 by dontsovcmc