dontsovcmc/waterius #319: HA, mqtt: перепутаны каналы холодной и горячей воды.

vzagorovskiy, 2024-05-04:
При настройке ватериуса устанавливаем настройки для холодной воды, для горячей указываем - отключен

В итоге в интерфейсе HA создается сенсоры только для ГОРЯЧЕЙ воды, в которых все по нулям.

Если не отключать горячую воду при настройке то все создается корректно.

vzagorovskiy, 2024-05-04:
в автоконфиге создаются только ch0, imp0, delta0 и т.д.
также педаются данные  только с индексом 0

Подозреваю путаница возникает из-за того что в интерфейсе настройки ватериуса первым идет холодная вода (синие контакты), а в хранении это данные с индексом 1.

dontsovcmc, 2024-05-04:
> в автоконфиге создаются только ch0, imp0, delta0 и т.д. также педаются данные только с индексом 0
> 
> Подозреваю путаница возникает из-за того что в интерфейсе настройки ватериуса первым идет холодная вода (синие контакты), а в хранении это данные с индексом 1.

ух. А какая версия прошивки?

vzagorovskiy, 2024-05-04:
Версия программного обеспечения: 1.1.2.32

vzagorovskiy, 2024-05-04:
https://github.com/dontsovcmc/waterius/releases/tag/1.1.0

dontsovcmc, 2026-08-19:
Разобрались: это чинилось в 1.1.4.

Коммит 019fffd («discovery channel if used», 31.05.2024) добавил проверку `channel_is_work()` перед публикацией сенсоров канала. До него `publish_discovery` публиковал сущности по `counterN_name` независимо от того, включён вход или нет.

Отсюда и путаница с холодной/горячей. Значения по умолчанию:

```c
uint8_t counter0_name = CounterName::WATER_HOT;    // красный вход
uint8_t counter1_name = CounterName::WATER_COLD;   // синий вход
```

Если настроить холодную воду (синий вход) и отключить горячую, имя `counter0_name` остаётся `WATER_HOT` — и в HA приезжали сенсоры горячей воды с нулями, потому что вход отключён.

Сейчас у выключенного входа публикуется только выбор типа входа, сенсоров нет. Обе каналы (`ch0` и `ch1`) есть и в discovery, и в payload.

В issue указана версия 1.1.2.32 — на две версии раньше фикса. Закрываю; если на актуальной прошивке (2.0.4x) симптом повторится, откроем заново.

closed 2026-08-19 by dontsovcmc