dontsovcmc/waterius #320: HA, mqtt: если отключить передачу данных на waterius.ru,то в mqtt данные тоже не отправляются

vzagorovskiy, 2024-05-04:
Если при настройке отключить передачу данных на waterius.ru,то в mqtt данные тоже не отправляются.
При этом топики  для autodiscovery создаются.

dontsovcmc, 2024-05-14:
Криминалити =(. Проверил код - сходу не вижу проблем... А можете лог прислать при сохранении страницы настройки отправки?

dontsovcmc, 2026-08-19:
Проверил код — воспроизвести не получается ни на текущей версии, ни на той, где писали issue.

Отправка в MQTT никогда не зависела от галочки waterius.ru. В 1.1.2 (`main.cpp`) ветки идут независимо:

```c
#ifndef WATERIUS_RU_DISABLED
    if (send_waterius(sett, json_data)) { ... }
#endif
#ifndef MQTT_DISABLED
    if (is_mqtt(sett)) { if (send_mqtt(...)) { ... } }
#endif
```

`is_mqtt()` смотрит только на `mqtt_on` и непустой `mqtt_host`. В `send_mqtt` настроек сайта тоже нет. Сейчас то же самое в `senders/send_data.cpp`.

Закрываю как невоспроизводимое. Если на актуальной прошивке (2.0.4x) симптом повторяется — переоткройте с логом из режима настройки, там видно, какие транспорты отработали.

closed 2026-08-19 by dontsovcmc