dontsovcmc/waterius #322: Не могу прошить FS на esp-01S

shax-xxx, 2024-05-14:
Здравствуйте.
Когда-то давно приобрёл ватериус. Версию прошивки сказать не могу но сам счётчик купил в сентябре 22г.
Внутри esp-01S.
Подключил, настроил отправку в mqtt один раз в час, а от него в заббикс графики там всякие и тд.
Батареек хватало примерно на полтора месяца. И надоело мне их менять так часто - выставил частоту отправки раз в сутки - а расход батареек остался прежний.
Решил прошить на новую версию. 
Есть у меня вот такой программатор: [](https://www.yourduino.ru/product/programmator-dlya-wifi-moduley-esp-0101s-na-baze-ch340g)

GPIO0 на землю и собственно через команды `python3 -m esptool ...` пытаюсь прошить.

```
/h/s/D/waterius # python3 -m esptool --port /dev/ttyUSB0 --baud 115200 write_flash --flash_freq 40m --flash_size 1MB --flash_mode dout 0x0 esp01_1m-1.1.2.bin 0xbb000 esp01_1m-1.1.2-fs.bin

esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:91:51:4f:99:e6
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00098fff...
Flash will be erased from 0x000bb000 to 0x000fafff...
Flash params set to 0x0320
Compressed 624160 bytes to 428061...
Wrote 624160 bytes (428061 compressed) at 0x00000000 in 57.3 seconds (effective 87.1 kbit/s)...
Hash of data verified.
Compressed 262144 bytes to 75303...
Writing at 0x000e7aa8... (100 %)Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/esptool/__init__.py", line 1139, in _main
    main()
  File "/usr/local/lib/python3.10/dist-packages/esptool/__init__.py", line 923, in main
    operation_func(esp, args)
  File "/usr/local/lib/python3.10/dist-packages/esptool/cmds.py", line 598, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "/usr/local/lib/python3.10/dist-packages/esptool/loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/esptool/loader.py", line 1077, in flash_defl_block
    self.check_command(
  File "/usr/local/lib/python3.10/dist-packages/esptool/loader.py", line 467, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/usr/local/lib/python3.10/dist-packages/esptool/loader.py", line 436, in command
    p = self.read()
  File "/usr/local/lib/python3.10/dist-packages/esptool/loader.py", line 369, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

```
Затирание через `python -m esptool --chip esp8266 --port COM3 --after no_reset erase_flash`  эффекта не даёт.

Если прошивать отдельно прошивку - всё хорошо.
Если прошивать отдельно fs - The chip stopped responding.

Пробовал через VSCode с platformio
Прошивка компилируется и заливается без проблем.
File system тоже собирается без проблем, но на этапе 'Upload Filesystem Image' :

```

post_buildfs([".pio/build/esp01_1m/littlefs.bin"], ["data"])
Move BIN files of filesystem version=1.1.0
/home/shax/Downloads/waterius/waterius/ESP8266/.pio/build/esp01_1m/littlefs.bin ->
/home/shax/Downloads/waterius/waterius/ESP8266/esp01_1m-1.1.0-fs.bin
Looking for upload port...
Using manually specified: /dev/ttyUSB0
Uploading .pio/build/esp01_1m/littlefs.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:91:51:4f:99:e6
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 262144 bytes to 74905...
Writing at 0x000bb000... (20 %)
Writing at 0x000bf000... (40 %)
Writing at 0x000c3000... (60 %)
Writing at 0x000c7000... (80 %)
Writing at 0x000cb000... (100 %)
Wrote 262144 bytes (74905 compressed) at 0x000bb000 in 17.2 seconds (effective 121.9 kbit/s)...

A fatal error occurred: Timed out waiting for packet header

```

Пробовал на своём модуле ESP-01 (без S) - там всё прошивается без ошибок, но после неё модуль, вставленный в waterius никак себя не проявляет кроме мигания лампочкой и зависания при нажатии кнопки.

Подскажите пожалуйста - как это залить внутрь родного модуля, а то идеи закончились?

shax-xxx, 2024-05-14:
прошил пока последнюю без FS 0.11.10-beta

dontsovcmc, 2024-05-14:
1. Что-то не так с расходом батареи. При раз в сутки должен работать несколько лет. 

2. А если вместо dout указать qout ?
python3 -m esptool --port /dev/ttyUSB0 --baud 115200 write_flash --flash_freq 40m --flash_size 1MB --flash_mode qout 0x0 esp01_1m-1.1.2.bin 0xbb000 esp01_1m-1.1.2-fs.bin

3. А можете прислать обозначение чипа памяти на ESP-02s?

4. Ваша ESP-01 будет работать (резистор можно потом удалить).
Прошивку attiny тоже надо обновить до посл. версии, чтобы посл .версия ESP работала. Думаю в этом дело.- 3 раза мигание это ошибка чтения конфига из attiny.

shax-xxx, 2024-05-14:
1. Да. я тоже об этом читал у вас, но такой вот факт. Могу статистику показать с заббикса.

2. qout - ситуацию не меняет

3. Видно не очень, но лучше не смог сфотографировать: 
[image]

4.Вот оно как. Я считал, что attiny вместо ESP-01S, а оно оказывается совместно в одном устройстве. Сейчас разглядел маркировку. А как его обновить? оно распаяно на плате.
[image]

shax-xxx, 2024-05-16:
Странное дело теперь стало происходить: 
В mqtt данные прилетают каждые 10 сек. Хотя в настройках период в 1440мин.
Это на 0.11.10-beta
Батарейки проживут ещё меньше :(

dontsovcmc, 2024-05-16:
> Странное дело теперь стало происходить: В mqtt данные прилетают каждые 10 сек. Хотя в настройках период в 1440мин. Это на 0.11.10-beta Батарейки проживут ещё меньше :(

Что-то странное, реально.
Attiny обновляется другим программатором.
Тут народная инстркция https://bit.ly/waterius-manual

Так, я бы плясал от версии прошивки attiny. Потому что мы не делали совместимость с ESP разных версий.

shax-xxx, 2024-05-17:
Инструкция порадовала.
Так а если я прошью новой прошивкой attiny - то он перестанет работать вместе с  0.11.10-beta у ESP ?

closed 2024-05-26 by shax-xxx